ArduinoGetStarted/button #3: INPUT_PULLUP Should be optional

epiller, 2020-12-25:
Using "INPUT_PULLUP" as default makes this library somewhat unusable if you've already wired the button with a pull-down resistor. I suggest this should be given as an option when creating the object, for example "ezButton button1(pin, pullup)"

ArduinoGetStarted, 2020-12-27:
Thank you for your comment. 
I think, most of Arduino beginner does not use the external pull-up/down resistor. However, Your idea is great. I will update it in the next version to make the library more flexible. The INPUT_PULLUP still will be set as default if users do not set the pull-up-down option.

ArduinoGetStarted, 2020-12-28:
Your suggestion was implemented

epiller, 2020-12-28:
Awesome! Don't forget to create a release so it's pushed to Arduino Library Manager :) Cheers!

closed 2020-12-28 by epiller

Vaneste, 2021-04-22:
Hi,
Just tried to use INPUT instead of INPUT_PULLUP mode but it doesn't seem to change anything. Any clue ?
Thanks !

ArduinoGetStarted, 2021-04-22:
Dear @Vaneste 
Thank you for your interest in my library.
For INPUT mode, it was added recently. I have not tested it. I have also not submit to Arduino Library Manager. I planed to update more for external Pull-up/down resistor before testing and submitting to Arduino Library manager.
I think INPUT mode will works if you download directly from Github and use an External pull-up resistor. If you do not use the external pull-up resistor, It may does not work because of the floating input

ArduinoGetStarted, 2024-08-09:
We just released new version 1.0.5 that supports internal pull-up/pull-down, external pull-up/pull-down.

Note:
* Some Arduino boards do not support `INTERNAL_PULLDOWN`
* ESP32 board: Pins GPIO34, GPIO35, GPIO36(VP) and GPIO39(VN) does not support `INTERNAL_PULLDOWN` and `INTERNAL_PULLUP`.
* It is good to read boards manual to check if a pin support `INTERNAL_PULLDOWN` and `INTERNAL_PULLUP`.