ArduinoGetStarted/button #7: Library does not support buttons wil pull down resistors

Divino-Fire, 2021-09-08:
I have a working project, it's stressful and errors prone for me to open it and change the button from which had a pull-down resistor to a pull-up resistor. I would like to just program the buttons to support debounce.

ArduinoGetStarted, 2021-09-09:
Dear  Divino-Fire,
The library has not supported pull-down resistor yet. We planed to add it in the next release. But we have not had a specific release date yet.

homonto, 2022-11-09:
I am 100% for inverted buttons as well to be added. Btw I am an electrician with hundred years of experience, and since I was in school, the ON was always when something is HIGH, so I always design my projects to be LOW = OFF, HIGH = ON.
Anyway, free software so nothing to complain about ;-)

jimtran93, 2023-03-09:
HI @ArduinoGetStarted! Just wondering if the library has been updated to support pull-down resistors. I am trying to use this on the Adafruit ESP32-S2 Reverse TFT that has two buttons that are normally low, and one button that is normally high.

hselomein, 2023-11-28:
Hi I am also wondering if there is pulldown support yet. I hac a project that uses 3 buttons in a pulldown configuration. in 3 GPIOs that do not have internal pull up or down resistors

ArduinoGetStarted, 2023-11-28:
Hello,

Thank you for your interest in the ezButton library.

Unfortunately, until now, I still have no time to update it to support the
pulldown. I am sorry for this inconvenience!

Have a nice day!
Best regards.

homonto, 2023-11-28:
> Hi I am also wondering if there is pulldown support yet. I hac a project that uses 3 buttons in a pulldown configuration. in 3 GPIOs that do not have internal pull up or down resistors

hi @hselomein 
use this one: [spacehuhn/SimpleButton](https://github.com/spacehuhn/SimpleButton) - it has full config including pullup and down

drf5n, 2024-08-08:
If you use pulldown resistors, you can make use of these bits:

https://github.com/ArduinoGetStarted/button/blob/2b2d1200d2bc0c7b28998ba25a32f825175bc3a5/src/ezButton.h#L57

https://github.com/ArduinoGetStarted/button/blob/2b2d1200d2bc0c7b28998ba25a32f825175bc3a5/src/ezButton.cpp#L34-L43

and write:

```
ezButton button(7,INPUT); // rely on external pulldown resistor
```
or if you are on a device with INPUT_PULLDOWN you can:

```
ezButton button(7,INPUT_PULLDOWN); // use internal pulldown resistor
```