ArduinoGetStarted/button #8: Library not working properly in a big project. What can hinder it ?

GeorgeFlorian, 2021-12-23:
Board: ESP32-EVB
GPIO: 15

I am trying to implement this in one of my projects and I am testing a PIN by putting it to LOW repeatedly using my hand and a wire.
But after a couple of seconds it registers tens of button presses at once.
```
The button is released
The button is pressed
The button is released
The button is pressed
The button is released
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is pressed
The button is released
```

My code contains a big ESPAsyncWebServer inside `void setup(){}`, and some routine handlers inside `void loop(){}`, one of witch should be buttons debounce.
I am using `delay(1)` inside `void loop(){}` because otherwise the WDT gets triggered after some `void loop(){}` cycles.

I have used the basic example with one button debounce and it worked great even under stress.

What can make it not work inside my project ?

ArduinoGetStarted, 2021-12-27:
Dear @GeorgeFlorian ,
I am sorry for the late reply.
GIOP15 PIN is internally connected to a touch sensor. It may cause a problem. Please try another pin such as GIOP16

GeorgeFlorian, 2021-12-27:
@ArduinoGetStarted
I am using the ESP32-EVB board, which has a limited number of GPIO:  13, 14, 15, 16, 32, 33, 34.
All the pins are used for something in my project and I cannot change them.
Unfortunately, switching pins is not an option for me.

closed 2022-01-21 by ArduinoGetStarted