ArduinoGetStarted/button #12: Conflict with SD.open within loop.

SensLmaup, 2022-03-07:
Having issues opening a file and writing to the file while in the loop. Everything works until it's put into the loops' code.

Using this code:

```
#include` <SPI.h>
#include <SD.h>
#define SD_CSpin 10
File myFile;

#include <Wire.h>   // I2C library
#include<I2CPart.h> // arbitrary part header

I2CPart part;
#include <ezButton.h>
#define LOOP_STATE_STOPPED 0
#define LOOP_STATE_STARTED 1
int buttonPin1=7;// usually 7
ezButton button(buttonPin1);  // create ezButton object that attach to digital pin;
int loopState = LOOP_STATE_STOPPED;

void setup(){
  // Enable serial
  Serial.begin(9600);

 // Enable I2C
  Wire.begin();
 bool ok= part.begin();
  Serial.println(ok ? "Part initialized" : "ERROR initializing part");

 //SD set up (initialation)
  if (!SD.begin(SD_CSpin)) {
    Serial.println(F("initialization failed!"));
    while (1);
  }
  Serial.println("initialization done.");

  // For ezButton use
  button.setDebounceTime(1000); // set debounce time to 1 second. (Hold for 1 second to start and stop)
}

void loop() {
  button.loop(); // MUST call the loop() function first

  if (button.isPressed()) {
    if (loopState == LOOP_STATE_STOPPED){
      loopState = LOOP_STATE_STARTED;
	  
	}else { // if(loopState == LOOP_STATE_STARTED)
      loopState = LOOP_STATE_STOPPED;
	  stop_actions();
	  }
  }

  if (loopState == LOOP_STATE_STARTED) {
	// Call the callback every PERIOD_MS
	meas_action(); // uses SD.open to print measurement data

  } 
}
```

emrifan6, 2022-08-15:
I have the same issue, hope the issue is resolved soon

ArduinoGetStarted, 2022-08-16:
@SensLmaup I am sorry for the late reply. Your code contains some hidden function that is potentially a blocking function. Please check how long the hidden function takes. If it takes long, ezButton or any other libraries that used timestamp may not work

@emrifan6 , Please check the above. Additionally, please make sure that your code does not use any [delay()](https://arduinogetstarted.com/reference/arduino-delay) function