ArduinoGetStarted/button #18: ESP32Cam only pin 4 working

borki74, 2023-01-23:
Hi,
I'm using the library with an ESP32Cam, and it works on pin 4, bot not on any other buttons.
Please help,

Matt

ArduinoGetStarted, 2023-02-03:
Dear @borki74 

I am sorry for the late reply.
I have not tested with ESP32Cam yet. I will test it when I have hardware and time.
If you find anything related to this, please share here.
Thank you

bertenvdb, 2023-04-20:
> Hi, I'm using the library with an ESP32Cam, and it works on pin 4, bot not on any other buttons. Please help,
> 
> Matt

I've had a similar problem with my AZDelivery D1 Mini ESP32. Two pins would work, the other ones didn't. I noticed ezButton has 2 constructors:
```
ezButton(int pin);
ezButton(int pin, int mode);
```
Looking at the implementation: 
`ezButton::ezButton(int pin): ezButton(pin, INPUT_PULLUP) {}; `

So I double checked the datasheet of my board and apparently the pins that wouldn't work don't have an internal pullup resistor. My problem was solved by using INPUT instead of the default INPUT_PULLUP.