ArduinoGetStarted/button #21: Nano ESP32 and INPUT_PULLUP

flyrdm, 2023-08-07:
I am using the library with a Nano ESP32 and the pullups on the pins D6 and D7 seem to "float" unless I add an explicit INPUT_PULLUP  on these pins inside of the setup() function. If add the INPUT_PULLUP before the setup() function, the problem is still there. By "float" I mean that the value will switch between HIGH and LOW
at random times. I was wondering if something in the ESP32 architecture might ignore the default pullups in the EZbutton constructor? Btw - thanks for the library, it saved me a lot of headaches with having to do my own debounce, etc.

ArduinoGetStarted, 2023-08-08:
Dear @flyrdm ,
Thank you for using our library.

Could you please try this constructor:  ezButton(pin, INPUT_PULLUP) ?
And let us know the result?
We do not have  Nano ESP32 to test here. So we need your help!

flyrdm, 2023-08-08:
I tried your suggestion of enabling it in the constructor and it had the same erratic/floating behavior. I then re-enabled the explicit pullup in the setup() func and it worked normally again. I have spares on hand and can send you a Nano ESP32 for help with this and as gratitude for your lib.

ArduinoGetStarted, 2023-08-08:
Dear @flyrdm 

Thank you.

I rechecked and realized that pin D6 and D7 of ESP32 are connected to the Flash Memory. It should not be used for GPIO. 
If you test with another pins (e.g. D4) and if it works, the problem is not in Library.

Could you please test and confirm it ?

epochjh, 2023-08-10:
The Nano ESP32 is ESP32-S3 based I believe. From this doc:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html

SPI0/1: GPIO26-32 are usually used for SPI flash and PSRAM and not recommended for other uses. 

So I don't think flash interferes but I am still going to do your experiment with other pins to see if I can find some that work and some that don't.

ArduinoGetStarted, 2024-08-09:
We just released new version 1.0.5 that supports internal pull-up/pull-down, external pull-up/pull-down.

Note:
* Some Arduino boards do not support `INTERNAL_PULLDOWN`
* ESP32 board: Pins GPIO34, GPIO35, GPIO36(VP) and GPIO39(VN) does not support `INTERNAL_PULLDOWN` and `INTERNAL_PULLUP`.
* It is good to read boards manual to check if a pin support `INTERNAL_PULLDOWN` and `INTERNAL_PULLUP`.

closed 2024-08-09 by ArduinoGetStarted