ArduinoGetStarted/button #23: 1.0.5 fails to compile on any device without INPUT_PULLDOWN

nomakewan, 2024-08-09:
As of 1.0.5, due to `INTERNAL_PULLDOWN` being strictly defined to `INPUT_PULLDOWN`, if `INPUT_PULLDOWN` does not exist on the platform which is being compiled for (for example, Arduino Mega 2560 or Arduino Nano), the compile will fail. Here is a log to demonstrate:

```
  In file included from /home/runner/Arduino/libraries/ezButton/src/ezButton.cpp:32:0:
  /home/runner/Arduino/libraries/ezButton/src/ezButton.cpp: In constructor 'ezButton::ezButton(int, int)':
  /home/runner/Arduino/libraries/ezButton/src/ezButton.h:43:27: error: 'INPUT_PULLDOWN' was not declared in this scope
   #define INTERNAL_PULLDOWN INPUT_PULLDOWN
                             ^
  /home/runner/Arduino/libraries/ezButton/src/ezButton.cpp:42:41: note: in expansion of macro 'INTERNAL_PULLDOWN'
    if (mode == INTERNAL_PULLUP || mode == INTERNAL_PULLDOWN) {
                                           ^~~~~~~~~~~~~~~~~
  /home/runner/Arduino/libraries/ezButton/src/ezButton.h:43:27: note: suggested alternative: 'INPUT_PULLUP'
   #define INTERNAL_PULLDOWN INPUT_PULLDOWN
                             ^
  /home/runner/Arduino/libraries/ezButton/src/ezButton.cpp:42:41: note: in expansion of macro 'INTERNAL_PULLDOWN'
    if (mode == INTERNAL_PULLUP || mode == INTERNAL_PULLDOWN) {
                                           ^~~~~~~~~~~~~~~~~
  /home/runner/Arduino/libraries/ezButton/src/ezButton.h:43:27: error: 'INPUT_PULLDOWN' was not declared in this scope
   #define INTERNAL_PULLDOWN INPUT_PULLDOWN
                             ^
  /home/runner/Arduino/libraries/ezButton/src/ezButton.cpp:49:14: note: in expansion of macro 'INTERNAL_PULLDOWN'
    if (mode == INTERNAL_PULLDOWN || mode == EXTERNAL_PULLDOWN) {
                ^~~~~~~~~~~~~~~~~
  /home/runner/Arduino/libraries/ezButton/src/ezButton.h:43:27: note: suggested alternative: 'INPUT_PULLUP'
   #define INTERNAL_PULLDOWN INPUT_PULLDOWN
                             ^
  /home/runner/Arduino/libraries/ezButton/src/ezButton.cpp:49:14: note: in expansion of macro 'INTERNAL_PULLDOWN'
    if (mode == INTERNAL_PULLDOWN || mode == EXTERNAL_PULLDOWN) {
                ^~~~~~~~~~~~~~~~~
```

Some sort of check needs to be in place here to prevent being able to define INTERNAL_PULLDOWN. Something like:

```
#ifdef INPUT_PULLDOWN
#define INTERNAL_PULLDOWN INPUT_PULLDOWN
#else
#define INTERNAL_PULLDOWN INPUT_PULLUP
#endif
```

Above is a quick and dirty example that would "fix" the problem by just assigning `INPUT_PULLUP` to `INTERNAL_PULLDOWN` if the selected platform lacks `INPUT_PULLDOWN`. Obviously this isn't the best solution since you're now creating a misconception (that `INTERNAL_PULLDOWN` is actually using a pulldown), but it would make it so no changes to the CPP file would be required.

ArduinoGetStarted, 2024-08-09:
Thank you for your report.

I compiled with Uno R4 and no error. Problem happen with Arduino Uno R3.
I am going to change it as below:
```
#ifdef INPUT_PULLDOWN
#define INTERNAL_PULLDOWN INPUT_PULLDOWN
#else
#define INTERNAL_PULLDOWN INPUT
#endif
```

Users will take responsibility to check if hardware supports INPUT_PULLDOWN

ArduinoGetStarted, 2024-08-09:
I just released 1.0.6, which changed the constant to the below:
```
#define INTERNAL_PULLUP  INPUT_PULLUP
#ifdef INPUT_PULLDOWN
#define INTERNAL_PULLDOWN INPUT_PULLDOWN
#else
#define INTERNAL_PULLDOWN INPUT
#endif

#define EXTERNAL_PULLUP   0xFE
#define EXTERNAL_PULLDOWN 0xFF
```

nomakewan, 2024-08-09:
Just tested those changes and compile operations no longer fail! Just need to wait for 1.0.6 to get noticed by the library manager now. Thank you so much!

closed 2024-08-09 by nomakewan