ArduinoGetStarted/button #24: Button initial state is wrong if it is pressed when arduino starts

jeanfabre, 2024-09-15:
If the button is pressed when you start the program, previousSteadyState is wrong, and you first need to wait for the button to be released for the ezbutton to start reflecting the real state.

Solution: previousSteadyState should be set to -1 and during the setup or the first loop detected as not yet HIGH or LOW and dealt with accordingly ( setting lastSteadyState to the opposite for the rest of the logic to work).

spaelectronics, 2024-09-19:
I'm having this same issue.

Can you please explain better how to apply your fix?

jeanfabre, 2024-09-19:
Hi,

I am planning on making a making a fork with the fix in it. but for now, this is the new function you need to add to the ezbutton.cpp

`void ezButton::refreshState(void)
{
	previousSteadyState = digitalRead(btnPin);
	if (previousSteadyState == HIGH)
	{
		lastSteadyState = LOW;
	}
	else {
		lastSteadyState = HIGH;
	}
	lastFlickerableState = previousSteadyState;

	lastDebounceTime = 0;
}

and you call refreshState during your setup callback in your main code. "refreshState()" should really be renamed setup() for clarity

spaelectronics, 2024-09-19:
I did this, but it's still not registering that a button is pressed on startup.

Any suggestions?

spaelectronics, 2024-09-19:
I fixed it by changing 
`lastFlickerableState = previousSteadyState;` to `lastFlickerableState = lastSteadyState;`