ArduinoGetStarted/button #26: License for button and other libraries

gworroll, 2025-01-09:
I saw a license.txt in the led library, looks like the three clause BSD. 

Does the same license apply to other libraries, such as this one, that you have in your repos? I've got a project I'm working on and need clear licenses. If it's the same as the led library that will work for my purposes.

ArduinoGetStarted, 2025-01-10:
Yes, the license is applied for all library in my GitHub. Some library have not added the license.txt yet. It will be added in the next version. However, the source code file already included the license. For license in short: You can freely use it but need to include the credit (link) to my original work when you publish or redistribute it in the text form (source code), no requirement for binary form (binary code after compiled).

gworroll, 2025-01-11:
Awesome, thanks. I should have looked at the source not just the list of files. That will do well for my purposes. Great work, bouncing could be a real problem in my thermostat project, changing the setpoint by several degrees for each press would not be great and manually doing that for up and down would be some messy code.